elhep/GEM_AFE_128 #1: BOM changes

gkasprow, 2022-10-17:
MIC45116-1YMP -> LTM4622AEY#PBF; default
24AA64T-I/MC -> 24AA64T-E/MNY ; add footprint

CannaCardo, 2022-10-21:
24AA64T-I/MC / 24AA64T-E/MNY footprints are close to identical, from what i can tell we should be able to place them on top of each another. Is that ok?

24AA64T-I/MC             |  24AA64T-E/MNY
:-------------------------:|:-------------------------:
[image] | [image] 

[image]

gkasprow, 2022-10-21:
OK, but I'd add SOIC option as well. These chips were hard to source recently