elhep/GEM_AFE_128 #2: SC18IS602 is EOL

gkasprow, 2024-07-02:
replace with SC18IS606PWJ

aw-pl, 2025-03-04:
Remember to keep layout of use of GPIOs for each chip since there are specific arranged and the communication can break.
Necessary to leave on GPIO unconnected for SN74 chips, since it looks that SC18 **won't issue transfer** if no SlaveSelect line is indicated.

gkasprow, 2025-03-20:
@aw-pl make sure that original ICS were not purchased before replacing them

gkasprow, 2025-03-20:
I confirmed, please replace them

filipswit, 2025-03-30:
I replaced them, but since there are only 3 GPIOs, I connected **DAC_CSn** to GPIO1 in IC31 instead of GPIO3 in 32. Is it fine?

After modifications:
[image]

Before modifications:

[image]

gkasprow, 2025-03-31:
We would have to add 128 pullups, otherwise it doesn't make any sense.

filipswit, 2025-03-31:
Why 128? Can't we put one pullup on **DAC_RESETn** for all 8 DACs?

filipswit, 2025-04-02:
@aw-pl finaly connection looks like this

[image]

gkasprow, 2025-04-03:
[image]

gkasprow, 2025-04-03:
@filipswit connect VREFP to P3V3

aw-pl, 2025-04-04:
It is connected now.
@filipswit **Make a variant with R31 as DNP by default.**

[image]

filipswit, 2025-04-04:
Added variant

[image]

aw-pl, 2025-04-04:
Checked, OK