elhep/GEM_AFE_128 #5: Correct addressing on LM75 chips

aw-pl, 2024-09-22:
The schematic is the same as on FMC ADC, so the IC38 and IC39 (LM75) have the same addresses. Need to correct this.

filipswit, 2025-03-23:
Changed IC28 and IC33 addresses, A1 to high.

[image]

filipswit, 2025-04-03:
I2C map
EEPROM on AFE is in conflict with EEPROM on FMC wehen GA0 = GA1 = 1

[image]

aw-pl, 2025-04-03:
@filipswit  think we have also MAX ADCs on this I2C?

aw-pl, 2025-04-04:
The LM75 are 0x4B and 0x4A, please correct only on the yellow note, but this also doesn't overlap with other chips.
One FMC side addresses are correct for LM75.

filipswit, 2025-04-04:
You are right, I looked at power supply as a address pin: 

[image]