elhep/GEM_AFE_128 #9: AFE measurements

aw-pl, 2025-02-07:
# Test Report

**Tests:** AFE channel 0 – AFE-ADC_FMC1, AFCKU  

---

## ADC Settings

- **Mode:** "DTGC"
- **Power Mode:** "Low-power"
- **ADC Resolution:** 10 bit

---

## DAC Settings

- **DAC Range:** 50mV (0x0) to 2.6V (0xfff) – *check #8 *
- **DAC Model Equation:** `0.0006237 * A1 + 0.045`

### DAC Configuration Table

| DAC Setting | DAC [mV] | R4 [mV] | R6 [mV] | ILA_CH0 [dec] | Comments                              |
|-------------|----------|---------|---------|---------------|---------------------------------------|
| 0x0         | 45       | 715     | 1110    | 1731          |                                       |
| 0x100       | 205      | 740     | 1100    | 1730          |                                       |
| 0x200       | 364      | 740     | 1100    | 1730          | No change on R4                       |
| 0x400       | 684      | 740     | 1100    | 1740          | No change on R4                       |
| 0x450       | 734      | 753     | 1110    | 1800          | Minimum change on R4                  |
| 0x500       | 843      | 850     | 1210    | 2290          | (from 0x400-500 = ~200mV diff)          |
| 0x700       | 1163     | 1160    | 1540    | 3757          |                                       |
| 0x750       | 1213     | 1210    | 1600    | 3974          |                                       |
| 0x800       | 1322     | 1310    | 1720    | 4095          | ILA_CH0 – high limit reached          |
| 0xa00       | 1642     | 1340    | 1750    | 4095          |                                       |
| 0xe00       | 2280     | 1540    | 1750    | 4095          |                                       |
| 0xfff       | 2599     | 1840    | 1740    | 4095          |                                       |

---

**Notes:**

1. Values are Vavg from scope.
2. `DAC_mV` is not measured directly but calculated using the approximate equation.

**Efficient DAC Range:** 0x450 – 0x800 = approx. 588 mV  
**Vref for DAC:** 2.048V (not checked yet)

aw-pl, 2025-02-07:
[image]

aw-pl, 2025-02-14:
Some new behaviour:
The same DAC channel on another system startup is getting correctly DAC 0x700 -> R4 = 1.16V, but now on R6 is **2.16V**

aw-pl, 2025-02-14:
Normal board power consumption- FMC1, AFE and no GTP working - 2.35A @ 12V

aw-pl, 2025-02-14:
If all channels on DAC are set to 0x0, and only one, like ch0 is set to something else, this works like in the table above. 
However, enabling more channels leads to situation, when the R4 values normally changes, but the R6 voltage goes easily above the limit and isn't blocked any more to the 1.7V value ->
now setting 0x750 on all DAC gives R4=1.21 CH0 and R6= 1.79V.
However, the ADC input channel is not in saturation mode, but gives 1910 (DEC) output.
Setting all channels to 0x0, and only ch0 to 0x750 gives as previously 3955 (DEC)

It can be observed on the scope, that R6 on CH0 rises slightly when more and more channels are getting their values, for example changing from 0x0 to 0x750.

DAC tested -> dac@f (first DAC ch0-15)

aw-pl, 2025-02-14:
More tests, on ADC FMC1 channels [1, 3, 5, 7, 11] are having moreless constant value of 2000 (DEC). Other channels reacts OK on offset changes.

aw-pl, 2025-02-14:
spread in values (ignore channels from comment above):

[image]

aw-pl, 2025-02-14:
However, changing all channels to 0x500 (128ch) make some kind of effect that clocks in some saturation/latch-up state even working channels-> now everything is around 2000 DEC.

aw-pl, 2025-02-15:
Made tests of all OFFSETs settings, setting each 16 channels per chip to 0x550 - that should give some middle range of ADC.
On the plot in bold is indicated what DAC have been changed. By default all DACs are set to 0x0. 

Note the fluctuations of all channels.
Another note: it is possible by changing for example DAC5 to stuck the ADC channel 0 (completly different), that is getting value around 2000 DEC and does not have any signal and also by moving DAC0_CH0 doesnt improve this.

[image]

aw-pl, 2025-02-17:
Another observation:
during different startup dac values the channels can be latched-up?  for example using 0x550 for all dacs at powerup, the ch0 and ch2 gets stuck at 1900-2000 DEC values, and then the DAC can be set to 0xf00 and it doesn’t change the input channel.

aw-pl, 2025-02-17:
Additional note for the above comment:
In fact the kynar wire was connected to ch0 and then connected to probe of the scope. During today's tests this wire got disconnected (at R6, so after omamp). Maybe the difference of some minimum capacitances etc makes those channels blocked. Need to check this by resoldering the wire.

aw-pl, 2025-02-17:
| Field 	| Current setting | Final settings 	| Comment 	|
|-------	|---------	|-------	|---------	|
| MODE_SEL     	| External non-uniform mode        	|  Programmable fixed-gain mode     	|  Needs to be changed       	|
| INP_RES_SEL | 0000 = 50ohm        	|       	|  Do we change this? Table 134/135 and 8.3.2      	|
| DIS_ATTEN | 0 | | Enabled |
| MANUAL_GAIN_DTGC | 0 | | we need to configure this - check below comment |
| PWR_MODE | LOW_POWER | LOW_POWER? | check comment below|
| GENERAL_MODE | DTGC | DTGC | other option is CW but it is not for our use|

According to 8.3.7.3.1 about fixed gain and atten. and table 6:
We are working in lowe power mode, so we have current ranges:
0 to 63 is the setting for attenuator -> 0 - -8dB
64 to 319 for the LNA -> 20-51 dB

Maximum input swing support (at INP_SOURCE(1)) -> 400mVpp

The above can change if switched to High Power mode - if necessary, but possible to get INP_SOURCE to 800mVpp

aw-pl, 2025-02-19:
Updated the configuration of the ADC chip to the following reference one from tests:
        // DTGC settings based on tests:
        // DIS_ATTEN = True
        // LNA_HPF_DIS
        // LNA_HPF_PROG 150kHz
        // LPF_PROG 25 MHz
        // High power
        // gain = 1
        mode_sel = <0x02>; // 10 = Programmable fixed-gain mode
        inp_res_sel = <0x00>; // 50Ohm // not used
        dis_atten;
        dtcg_atten = <0>; // not used
        dtcg_gain = <0>;  // gain=1

        // LNA_HPF_DIS
        lna_hpf_dis;        
        // LNA_HPF_PROG 150kHz
        lna_hpf_prog = <150>;
        // LPF_PROG 25 MHz
        lpf_prog_hpm = <25>;

high power mode.

This have significant effect on the SN74LV595APWT chips, since now OE have effect on the CAL_OUT* lines that changes the voltage really 50mV - 3.3V

With new settings upcoming new channels readout plot with 0x550 offset setting

aw-pl, 2025-02-19:
Working in high power mode 12V@2.75A

aw-pl, 2025-02-20:
Checked status of `TR_EN `pins. 
They have internal pull-up. That means that the LNA HPF are **connected** to the INP pins by default. No need to configure this by registers (alternative way).

aw-pl, 2025-02-27:
| dac | adc_offset | adc_input 	| Comment 	|
|-------	|---------	|-------	|---------	|
|0x0| 0.94| 0.962|
| 0x100|0.94| 0.962|
|0x400| 0.94|0.963
|0x450| 0.960 | 0.95
|0x550 | 0.964 | 1.14|-  saturation of adc
0x650| 0.955 |1.33|

aw-pl, 2025-02-27:
**R40 moved to value 7kOhm to get ref voltage 1.2V instead of 1.024V**

aw-pl, 2025-02-27:
Starting the system with all dacs set to 0x450

aw-pl, 2025-02-27:
channel 2
| dac | adc_offset r27  | adc_input r26| Comment 	|
|-------	|---------	|-------	|---------	
|0x550| 1.07 | 1.25| 
|0x5b0 | 1.085| 1.32V | this is exact saturation level of adc

aw-pl, 2025-02-27:
After all dacs set to 0x750 (around 1.2V) on each channel, now the channels CH0 and CH2 have voltage around 2.0V, CH0 looks like it is really connected(?)

aw-pl, 2025-02-27:
on adc now on ch0 the minium voltage is 1.57. (r26) and r27=2.03V

aw-pl, 2025-02-28:
Scan of voltages based on new DACs configurtion
DAC offset on all channels the same settings
note - check if this is not working ch0 or ch2
| dac | adc_offset r27  | adc_input r26| Comment 	|
|-------	|---------	|-------	|---------	
0x0 | 0.9645 | 0.9611
0x300 | 0.9644 | 0.9611
0x400 | 0.9643| 0.9608
0x450|  0.9646 | 0.9609
0x500| 0.9757 | 0.9730
0x550 | 1.0075 | 1.0072
0x600 |1.1845 | 1.1881
0x650 | 1.2759 | 1.2801
0x700 | 1.4724 |1.4779
0x750 | 1.5624 | 1.5679
0x900 | 2.0521 | 2.0589
0xa00 | 2.2354 | 2.2418

the same measurements on second channel (check the numbers ch0 and ch2)
 | dac | adc_offset r27  | adc_input r26| Comment 	|
|-------	|---------	|-------	|---------	
0x0 | 0.9641 | 0.9639
0x300 | 0.9641 | 0.9640
0x400 | - | not measured
0x450|  0.9643 | 0.9641
0x500| 0.9756 | 0.9751
0x550 | 1.0077 | 1.0065
0x600 | 1.1852 | 1.1823
0x650 | 1.2770 | 1.2737
0x700 | 1.4737 | 1.4695
0x750 | 1.5632 | 1.5582
0x900 | 2.0548 | 2.0477
0xa00 | 2.2351 | 2.1995

Note: dac ch0 -> settings are correct on the output: 0x700 gives around 890mV (should give 875mV)

aw-pl, 2025-03-01:
Check P3V3A = 3.40V OK

aw-pl, 2025-03-01:
The CAL_ON signal works OK , changes state 1/0 -> changes state by IC31. 
However for the moment no difference on outputs of SN74 - checking further

aw-pl, 2025-03-01:
**WARNING**! Problem with ground/ref potentials! To be checked

aw-pl, 2025-03-01:
After adding slave select on unused gpio of  IC31 the SPI for SN74 chips works. Checking further

aw-pl, 2025-03-01:
SPI data lines CLK, SRDAT, SRDAT_OUT working correctly, updated driver codes.
Now checking cal_out

aw-pl, 2025-03-01:
CAL_ON, CAL_OUT and whole SN74 programming registers works correctly. 
Checking now FE channel after opamp IC1C

aw-pl, 2025-03-01:
Got first cal pulse!!! Channel [2], offset 0x550, 
Pulse: 117ns, amplitude 740mV!

aw-pl, 2025-03-03:
Cal pulse test
CH[2] - DAC setting 0x550
Pulse on R6 (AFE): 
amplitude: 790mV; 
offset: 1.26V; 
Peak: 2.05
time: 117ns

R26 value:
amplitude: 630mV; 150mV difference on cable 
offset: 1.33V; 
Peak 1.96V:
time: 117 ns
**OSCILLATIONS!**

aw-pl, 2025-03-03:
Checking now on chipscope, this channel got for the first time **1464** DEC value. (so now below this **2000DEC**).

First pulse on chipscope!
Offset: 1460 DEC; peak 2400 DEC. Time: 7 samples (50MHz) = around 140ns
[image]

aw-pl, 2025-03-03:
Ok, now we have a bit different situation, like here:
This is after fresh startup, and all the channels are now again stuck at 2000 DEC offset...
Reconnected multimeter to the same probe as probe from oscilloscope, does it influence? Different grounds (?). Offsets set up globally to 0x550 from 0x750.

[image]

aw-pl, 2025-03-03:
Now suddenly more channels responded to the cal pulse (after replacement of one broken r6 on ch[0] afe):

[image]

[image]

aw-pl, 2025-03-03:
after reconnecting probes ch[2] again with good signal (?)

aw-pl, 2025-04-02:
[image]

aw-pl, 2025-04-02:
[image]

aw-pl, 2025-04-02:
[image]

aw-pl, 2025-04-02:
[image]

aw-pl, 2025-04-02:
[image]

aw-pl, 2025-04-02:
Two last channels on this chip not working
[image]

aw-pl, 2025-04-02:
[image]

aw-pl, 2025-04-02:
[image]

aw-pl, 2025-04-02:
[image]

aw-pl, 2025-04-02:
[image]

aw-pl, 2025-04-02:
[image]

aw-pl, 2025-04-02:
[image]

aw-pl, 2025-04-02:
[image]

aw-pl, 2025-04-02:
Those channels have some issues:
oscillating

[image]

will reupload firmware
[image]

aw-pl, 2025-04-02:
[image]

above one more channel

reupload firmwar does not change anything

aw-pl, 2025-04-02:
- [ ] Check lines with oscillations
- [ ] check adc1- ch30/31 about routing (is it routing issue)

filipswit, 2025-04-03:
@aw-pl please check on another FMC card.

aw-pl, 2025-04-03:
Checked, however I cannot get lanes calibrated for this chip, the 30/31 channels are similar to serial transmission issue (training only rather).
For the pulses it is unknown, however the same lines are not working.
Line 30 is stuck at some level, however remember that lanes are not trained here well. If possible I would suggest to improve the layout on analog and digital side for this chip (serial links). 

[image]

[image]

aw-pl, 2025-04-04:
Ok, changed the IDELAY value for the whole ADC[1] chip to 0x05, it has corrected the channels issues. 
30/31 is normally working
18/19 no issue with the impulses (due to digital transmission)

Note: Test were done on bit different version of the firmware, so probably not the whole chip was configured as previously. However doesn't change anything for this test

[image]

[image]

[image]

[image]

aw-pl, 2025-04-04:
Checked also second FMC board, adjusting globally the whole chip makes the readings correct