elhep/GEM_AFE_128 #10: SN74LV595 CAL_PULSE outputs

aw-pl, 2025-02-17:
The write/read to the chips despite or SRCLK and RCLK connected together works and is possible to get readback from all the chips.

However, no change on the output is present. The change of CAL_ON (nOE on chip) makes the change on output voltage by 10mv only, from around 39mV to 49mV.

This needs to be checked.

gkasprow, 2025-02-17:
that can make sense, we have very small calibrator capacitance, try to increase one by 10x

aw-pl, 2025-02-19:
This is solved now, in fact inproper configuration of the ADC chips was causing this effect, that nOE didn't be able to control output lines CAL_OUT*. 
Now it is possible to control those lines, so transitions 0->1 and 1->0, but there is no pulse on the R4.

Will change now the capacitor C3 to 33pF

aw-pl, 2025-03-04:
No need to change capacitor, we are already getting correct pulses. Check #9 

SOLVED (not closing due to design corrections)