elhep/GEM_AFE_128 #11: SN74LV595 reconnect control signals

aw-pl, 2025-02-25:
Connect RCLK to FPGA directly
nOE either connect to fixed value (GND) or leave it at GPIO of SC18IS602

filipswit, 2025-03-29:
@aw-pl Should **RCLK** be connected to **CAL_ON** from FMC?

Connected nOE SN74LV595 to GND
[image]

gkasprow, 2025-03-29:
Yes

Pozdrawiam / Best regards
G.Kasprowicz

image.png (view on web)<https://github.com/user-attachments/assets/74b80cb8-1106-4f8f-b252-a267cb266cc0>

image.png (view on web)<https://github.com/user-attachments/assets/74b80cb8-1106-4f8f-b252-a267cb266cc0>

aw-pl, 2025-03-31:
Is it possible (low-cost modification) to connect nOE to AFE connector and control it by this I2C-GPIO multiplexer? Could be quite good options.

filipswit, 2025-03-31:
Looks feasible.
We need 3.3 logic, but I2C expander works on 1.8, so I think open drain solution for OE signal should be fine.

[image]

gkasprow, 2025-03-31:
@aw-pl what's the purpose of controlling the OEn singnal?

aw-pl, 2025-04-03:
I was thinking to output all of the channels at once (0/1 switching), connecting clock-type signal from FPGA, so we could simulate count-intensive pulses (by making simple clock generator in FPGA). So we would progam the shift register with 0's and change nOE to either have pull-up or GND. But as you mentioned that would require pull-ups everywhere.

So I think it is not necessary to do this and too complicated.

gkasprow, 2025-04-04:
it wont work!
we don't have pullup resistors.
we could only generate 0->1 sequence, not 1>0 because we have only pulldown

[image]