elhep/GEM_AFE_128 #13: connect unused signals to GND

gkasprow, 2025-03-04:
leave only 2 unconnected with test points

gkasprow, 2025-03-19:
I mean the AFE - ADC connector, leave 2 unconnected

filipswit, 2025-03-29:
Added 2 test points, CAL_ON signal and I leave IN_N[0..3] as placeholders.

[image]

filipswit, 2025-04-03:
THT pins on AFE-ADC connector are parialy connected to GND, Can I connect all of them to GND?

[image]

[image]

gkasprow, 2025-04-03:
Yes!

aw-pl, 2025-04-04:
@filipswit TP4 and TP5 shouldn't be connected to GND also?
[image]